raspberrypi/picamera2 #1300: [SUGGESTION] HailoDevice context manager for HailoModel

gh9869827, 2025-07-28:
## Is your feature request related to a problem? Please describe.

Not a blocking issue, but the current `HailoModel` implementation in `picamera2/devices/hailo/hailo.py` uses static class-level variables (`TARGET`, `TARGET_REF_COUNT`) to manage the shared Hailo virtual device. This approach does support using multiple models on a single device, but:

- It hides lifecycle management behind global state.
- It prevents users from explicitly controlling when the virtual device is created or released.

## Describe the solution you'd like

Refactor the code to:

- Introduce a `HailoDevice` class to explicitly manage the virtual device, with support for context-managed lifetime (`with HailoDevice() as device:`).
- Update `HailoModel` to take a `device: HailoDevice` as a parameter instead of relying on global variables.
- Optionally, use a `HailoModelConfig` NamedTuple to encapsulate model parameters (`hef_path`, `output_type`, `batch_size`).

This preserves all current functionality, including support for multiple models on the same device, while introducing a mechanism to explicitely control the lifecycle of the device using a `with` statement, independantly of the `HailoModel`s.

## Additional context

I've attached a proposed implementation for the mechanism I describe above. It includes header comments explaining the changes and preserves attribution to the original Picamera2 source.

I'd be happy to open a pull request if there's interest in this direction.

[hailo_py.txt](https://github.com/user-attachments/files/21478258/hailo_py.txt)

davidplowman, 2025-07-29:
Hi, and thanks for the suggestion. In principle I'm keen to accept contributions that will make things work better, just a couple of questions.

The existing code has `__enter__`/`__exit__` context manager methods, so are we replacing these? My impression with them was that they don't work correctly with multiple models - presumably this will fix that? We can "deprecate" those old methods with a warning, and indicate to users what the new replacement code would be.

We should obviously also add an example of this updated API to the repo so that it's clear how to use it. There's also a multiple model demo [here](https://github.com/raspberrypi/face_recogniser), perhaps we could update that to use this new context manager and demonstrate that it works correctly now?

I feel slightly that it might be worth making the changes that give us functional fixes first, and leave the cosmetic config changes for another time. I'm just always really nervous about changing more more stuff under our user's feet in one go than is absolutely necessary. Does that sound OK?

I'm also fretting slightly about the naming. In this new scheme, I kind of get the impression that the `HailoDevice` and `HailoModel` are named the wrong way round. There's really only one physical Hailo device, attached to the top of the Pi. Meanwhile, it can run multiple models. That makes for an annoying API change for users, however, so I'm not sure whether I would dare do it. Maybe the trick is to think of the `HailoDevice` more as a "virtual device`. Any thoughts?

Otherwise, yes please, I think we'd be very happy to consider a PR.

@naushir FYI

gh9869827, 2025-07-29:
Hello,

I took a closer look at how the static variables are used, and I think I see where things might not work correctly with multiple models.

In the `Hailo` class, I added three `print` statements to trace the execution.  
I used a simple test program that loads two Hailo models, enters a context block, and then exits.

With the current implementation, the `TARGET_REF_COUNT` variable is incremented only once, when the `VDevice` is allocated. Subsequent models don't increment the variable. However, when the context exits, the `TARGET_REF_COUNT` is decremented by each model, causing the `VDevice` to be deallocated prematurely.

I then used another simple test program that loads a single model twice consecutively.  
When the device is released, the `TARGET` variable is not reset to `None`, which causes the code to reuse a device that has already been deallocated.

I've added below 5 code/text snippets:
- The key logic from the `Hailo` class related to device allocation and deallocation
- A test program illustrating premature deallocation and its output using the current implementation
- A test program illustrating reuse of a released device and its output using the current implementation
- A proposed change (two lines)
- The output of the test programs using the proposed implementation

## Key logic from the Hailo class

```python
class Hailo:
    TARGET = None
    TARGET_REF_COUNT = 0

    def __init__(self, hef_path, batch_size=None, output_type='FLOAT32'):
        ...
        if Hailo.TARGET is None:
            print("Create VDevice()")
            Hailo.TARGET = VDevice(params)
            Hailo.TARGET_REF_COUNT += 1
        self.target = Hailo.TARGET
        ...
        print(f"Model loaded with TARGET_REF_COUNT={Hailo.TARGET_REF_COUNT}")

    def __enter__(self):
        return self

    def __exit__(self, exc_type, exc_val, exc_traceback):
        self.close()

    def close(self):
        del self.configured_infer_model
        Hailo.TARGET_REF_COUNT -= 1
        if Hailo.TARGET_REF_COUNT == 0:
            print("Release target VDevice")
            self.target.release()
```

## Test program illustrating premature deallocation

```python
from hailo import Hailo

m1 = Hailo("yolov8l.hef")
m2 = Hailo("yolov8l.hef")

with (m1, m2):
    pass
```

Output:
```
Create VDevice()
Model loaded with TARGET_REF_COUNT=1
Model loaded with TARGET_REF_COUNT=1
Release target VDevice
Segmentation fault
```

## Test program illustrating reuse of released device

```python
from hailo import Hailo

with Hailo("yolov8l.hef") as m1:
    pass

with Hailo("yolov8l.hef") as m1:
    pass
```

Output:
```
Create VDevice()
Model loaded with TARGET_REF_COUNT=1
Release target VDevice
Traceback (most recent call last):
  ...
AttributeError: 'NoneType' object has no attribute 'create_infer_model_from_file'
```

## Proposed change

Update the constructor so `TARGET_REF_COUNT` is incremented regardless of whether the device already exists, and update `close()` to reset `TARGET` when releasing the device.

```python
    def __init__(self, hef_path, batch_size=None, output_type='FLOAT32'):
        ...
        if Hailo.TARGET is None:
            print("Create VDevice()")
            Hailo.TARGET = VDevice(params)
        # Move REF_COUNT increment outside the condition
        Hailo.TARGET_REF_COUNT += 1
        self.target = Hailo.TARGET
        ...
        print(f"Model loaded with TARGET_REF_COUNT={Hailo.TARGET_REF_COUNT}")

    def close(self):
        del self.configured_infer_model
        Hailo.TARGET_REF_COUNT -= 1
        if Hailo.TARGET_REF_COUNT == 0:
            print("Release target VDevice")
            self.target.release()
            # reset `TARGET` when releasing the device
            Hailo.TARGET = None
```

## Output of test programs using the proposed implementation

```
Create VDevice()
Model loaded with TARGET_REF_COUNT=1
Model loaded with TARGET_REF_COUNT=2
Release target VDevice
```

```
Create VDevice()
Model loaded with TARGET_REF_COUNT=1
Release target VDevice
Create VDevice()
Model loaded with TARGET_REF_COUNT=1
Release target VDevice
```

## Additional comment

In addition, the resource allocation takes place in the constructor, whereas the resource deallocation takes place in the `__exit__` context manager. This means if models are created outside the `with` statement and reused, the device will not be reallocated before reuse.

```python
from hailo import Hailo

m1 = Hailo("yolov8l.hef")

with m1:
    pass

with m1:
    pass
```

gh9869827, 2025-07-30:
Hi, absolutely, I'd be happy to put this into a branch based on `next` and open a PR.