raspberrypi/picamera2 #1305: [BUG]hang of capture_file a jpg still or of capture_metadata after video recording in picam 0.3.30 on RPi 4

herbbetz, 2025-08-04:
For some weeks I have been trying picamera2 on bookworm, using a raspberry 4B and the simple rpicam v3. I need to use the same config for still images and video to avoid a time lag, when changing between the two. For this change I found that I completely have to do picam.stop() and picam.start() after a video recording, as otherwise the picam.capture_file() or picam.capture_metadata() will hang without any error message. The following code would not be working without the 'reset_camera(picam, camsetting, config)' : ``` import os
import subprocess
import time
import io
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder, Quality
from picamera2.outputs import FileOutput, CircularOutput
import libcamera
import importlib.metadata # only for picamera2 version printing, because there is no 'picamera2.__version__'

def capture_img(picam, dest):
    picam.capture_file(dest, name="main", format="jpeg") # lores YUV420 and jpeg are not compatible (error: Buffer has wrong number of dimensions (expected 2, got 3))

def reset_camera(picam, camsetting, config):
    start_ns = time.time_ns()
    picam.stop()
    picam.configure(config)
    picam.start()
    picam.set_controls(camsetting)
    print(f"reset_cam_time {(time.time_ns() - start_ns) // 1_000_000} ms")

def write_binVideo(newfname, binVideo):
    fname = newfname + '.h264'
    with open(fname, 'wb') as vfile:
        vfile.write(binVideo)

    mp4video = newfname + '.mp4'
    cmd = f'ffmpeg -y -framerate 24 -i "{fname}" -c copy "{mp4video}"'
    ffproc = subprocess.Popen(cmd, shell=True)
    ret = ffproc.wait() # await cmd completion
    if ret == 0: os.remove(fname)

def main():
    print(f'picam2 version {importlib.metadata.version("picamera2")}')

    camsetting = {
        "AeEnable": False,
        "AnalogueGain": 2.0,
        "ExposureTime": 50000,
        "AwbEnable": False,
        "ColourGains": (1.5, 1.2)
    }

    camera_transform = libcamera.Transform(hflip=1, vflip=1)

    with Picamera2() as picam:
        config = picam.create_video_configuration(
            main={"size": (1280, 960), "format": "RGB888"},
            lores={"size": (320, 240), "format": "YUV420"},
            transform=camera_transform
        )

        # encoder = H264Encoder() in send_movement()
        picam.configure(config)
        time.sleep(0.5)
        picam.start()
        picam.set_controls(camsetting)

        timestamp = round(time.time() * 1000)
    
        print("shooting a still")
        imgName = f"{timestamp}pre.jpg"
        capture_img(picam, imgName)

        print("posttrigger video")
        posttrigger = io.BytesIO()
        post_file_output = FileOutput(posttrigger)
        post_encoder = H264Encoder()
        picam.start_recording(post_encoder, post_file_output, quality=Quality.MEDIUM)
        time.sleep(6)
        picam.stop_recording()
        posttrigger.seek(0)
        postdata = posttrigger.getvalue()
        print(f"posttrigger_video size: {len(postdata)} bytes")
        write_binVideo(f"{timestamp}post", postdata)

        reset_camera(picam, camsetting, config)
        print("shooting a still")
        imgName = f"{timestamp}post.jpg"
        capture_img(picam, imgName)

        print("end")

if __name__ == "__main__":
    main() ```

herbbetz, 2025-08-04:
```
import os
import subprocess
import time
import io
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder, Quality
from picamera2.outputs import FileOutput, CircularOutput
import libcamera
import importlib.metadata # only for picamera2 version printing, because there is no 'picamera2.__version__'

def capture_img(picam, dest):
    picam.capture_file(dest, name="main", format="jpeg") # lores YUV420 and jpeg are not compatible (error: Buffer has wrong number of dimensions (expected 2, got 3))

def reset_camera(picam, camsetting, config):
    start_ns = time.time_ns()
    picam.stop()
    picam.configure(config)
    picam.start()
    picam.set_controls(camsetting)
    print(f"reset_cam_time {(time.time_ns() - start_ns) // 1_000_000} ms")

def write_binVideo(newfname, binVideo):
    fname = newfname + '.h264'
    with open(fname, 'wb') as vfile:
        vfile.write(binVideo)

    mp4video = newfname + '.mp4'
    cmd = f'ffmpeg -y -framerate 24 -i "{fname}" -c copy "{mp4video}"'
    ffproc = subprocess.Popen(cmd, shell=True)
    ret = ffproc.wait() # await cmd completion
    if ret == 0: os.remove(fname)

def main():
    print(f'picam2 version {importlib.metadata.version("picamera2")}')

    camsetting = {
        "AeEnable": False,
        "AnalogueGain": 2.0,
        "ExposureTime": 50000,
        "AwbEnable": False,
        "ColourGains": (1.5, 1.2)
    }

    camera_transform = libcamera.Transform(hflip=1, vflip=1)

    with Picamera2() as picam:
        config = picam.create_video_configuration(
            main={"size": (1280, 960), "format": "RGB888"},
            lores={"size": (320, 240), "format": "YUV420"},
            transform=camera_transform
        )

        # encoder = H264Encoder() in send_movement()
        picam.configure(config)
        time.sleep(0.5)
        picam.start()
        picam.set_controls(camsetting)

        timestamp = round(time.time() * 1000)
    
        print("shooting a still")
        imgName = f"{timestamp}pre.jpg"
        capture_img(picam, imgName)

        print("posttrigger video")
        posttrigger = io.BytesIO()
        post_file_output = FileOutput(posttrigger)
        post_encoder = H264Encoder()
        picam.start_recording(post_encoder, post_file_output, quality=Quality.MEDIUM)
        time.sleep(6)
        picam.stop_recording()
        posttrigger.seek(0)
        postdata = posttrigger.getvalue()
        print(f"posttrigger_video size: {len(postdata)} bytes")
        write_binVideo(f"{timestamp}post", postdata)

        reset_camera(picam, camsetting, config)
        print("shooting a still")
        imgName = f"{timestamp}post.jpg"
        capture_img(picam, imgName)

        print("end")

if __name__ == "__main__":
    main()
```

herbbetz, 2025-08-05:
Many thanks for the quick reply. Leaving the camera running, no matter if I use picam.stop_recording() or additionally picam.stop_encoder() makes the python script hang on the second still. With keyboard interrupt it shows:
 ```
...
shooting a still-2
^CTraceback (most recent call last):
  File "/home/pi/station3/camtest/videvents0.py", line 89, in <module>
    main()
  File "/home/pi/station3/camtest/videvents0.py", line 84, in main
    capture_img(picam, imgName)
  File "/home/pi/station3/camtest/videvents0.py", line 14, in capture_img
    picam.capture_file(dest, name="main", format="jpeg") # lores YUV420 and jpeg are not compatible (error: Buffer has wrong number of dimensions (expected 2, got 3))
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1571, in capture_file
    return self.dispatch_functions(functions, wait, signal_function)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1466, in dispatch_functions
    return job.get_result(timeout=timeout) if wait else job
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/job.py", line 87, in get_result
    return self._future.result(timeout=timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 451, in result
    self._condition.wait(timeout)
  File "/usr/lib/python3.11/threading.py", line 320, in wait
    waiter.acquire()
KeyboardInterrupt
```

davidplowman, 2025-08-05:
Hi, you would need to use `picam.stop_encoder()` _instead of_ `picam2.stop_recording()`. Don't use both. Sorry if I wasn't clear.

If the camera configuration is actually the same (which I think it is in the code you have posted), you don't even need to stop the camera at all before capturing the still.

herbbetz, 2025-08-05:
The following code is probably what you meant. And you were right: picam.stop_encoder() alone does not hang the following picam.capture_file() like picam.stop_recording() did. But then why is the .start_recording()/.stop_recording() useful at all? picamera2-manual.pdf does not mention this? Are the functions & attributes also documented elsewhere, apart from in the python REPL? Now this works:
```
        ...
        print("shooting a still-1")
        imgName = f"{timestamp}pre.jpg"
        capture_img(picam, imgName)

        post_encoder = H264Encoder()
        posttrigger = io.BytesIO()
        post_file_output = FileOutput(posttrigger)

        print("posttrigger video-1")
        picam.start_encoder(post_encoder, post_file_output, quality=Quality.MEDIUM)
        time.sleep(3)
        # picam.stop_recording() hangs the following picam.capture_file(). Also do not use together with picam.stop_encoder()!
        picam.stop_encoder() # replaces reset_camera() !!
        # reset_camera(picam, camsetting, config) # containing picam.stop()

        posttrigger.seek(0)
        postdata = posttrigger.getvalue()
        print(f"posttrigger_video size: {len(postdata)} bytes")
        write_binVideo(f"{timestamp}post-1", postdata)
        # clear ioBuffer for reuse:
        posttrigger.seek(0)
        posttrigger.truncate(0)

        print("shooting a still-2")
        imgName = f"{timestamp}post.jpg"
        capture_img(picam, imgName)

        print("posttrigger video-2")
        picam.start_encoder(post_encoder, post_file_output, quality=Quality.MEDIUM)
        time.sleep(3)
        picam.stop_encoder()
        posttrigger.seek(0)
        postdata = posttrigger.getvalue()
        print(f"posttrigger_video size: {len(postdata)} bytes")
        write_binVideo(f"{timestamp}post-2", postdata)

        print("end")

if __name__ == "__main__":
    main()
```

herbbetz, 2025-08-05:
P.S.: If circular_output is used, the hanging will not ensue:
```
    encoder = H264Encoder()
    circ_output = CircularOutput()
    picam.start_recording(encoder, c_output, quality=Quality.MEDIUM) # or better picam.start_encoder(...)
    time.sleep(5) # accumulate pretrigger frames

    outmem = io.BytesIO()
    circ_output.fileoutput = outmem
    circ_output.start()
    start_ns = time.time_ns()
    time.sleep(videodurate)
    circ_output.stop()
    outmem.seek(0)
    full_video = outmem.getvalue()
```

herbbetz, 2025-08-05:
Ah, yes, thanks again for the fast responses, I was confused, because it looked to me, that picam.start_recording() and picam.start_encoder() did the same thing, whereas for their stop versions, only picam.stop_recording() caused me problems. Funnily enough 'stop_encoder' gave me no search results in the picam2 manual with adobe reader, as I searched for a syntax list of all the picam2 functions. AI platforms often mention functions, that are not or no longer there and possible function arguments and return values need to be found in the extensive examples.

A final question, that is not an issue, but maybe you can give me a pointer. picamera2 can measure brightness and I am looking to adapt its camsettings accordingly. Could not find an example yet for this. Also I do not know, how to use github as a forum for such questions. The code I came up with looks like this:
```
def set_brightness(picam, camsetting, now):
    frame = picam.capture_array(name="main")
    avg_brightness = round(np.mean(frame[:, :, 0]))
    metadata = picam.capture_metadata()
    metalux = metadata.get("Lux", None) # metadata["Lux"]
    metalux = round(metalux) if metalux is not None else None
    '''avg_brightness is dependant on the exposure/gain of the camsetting, so if it hits light_level thresholds (i.e. avg_brightness <40 or <80 or <150 or else),
        it will alternate between two levels despite the same scene illumination.
        Therefore make an independant scene_brightness (a lux estimate, that could be calibrated by a scale factor to an external lux meter)'''
    scalefactor = 1e6
    exposure = camsetting["ExposureTime"]
    gain = camsetting["AnalogueGain"]

    sc_brightness = (avg_brightness * scalefactor) / (gain * exposure)
    sc_brightness = round(sc_brightness)

    if sc_brightness < luxThreshold[0]: # luxThreshold is a list: [600,2000,4000]
        light_level, luxcategory = "dark", 1
    elif sc_brightness < luxThreshold[1]:
        light_level, luxcategory = "dim", 2
    elif sc_brightness < luxThreshold[2]:
        light_level, luxcategory = "normal", 3
    else:
        light_level, luxcategory = "bright", 4

    if light_level != set_brightness.last_light_level:
        camsetting.update({
            "ExposureTime": { "dark": 50000, "dim": 30000, "normal": 15000, "bright": 6000 }[light_level], # 100–1,000,000 µs (e.g., 50,000 µs = 1/20s)
            "AnalogueGain": { "dark": 6.0, "dim": 3.0, "normal": 1.5, "bright": 1.0 }[light_level] # Set gain (e.g., 1.0 to 16.0 typical range)
        })
        picam.set_controls({
            "ExposureTime": camsetting["ExposureTime"],
            "AnalogueGain": camsetting["AnalogueGain"]
        })
        set_brightness.last_light_level = light_level

set_brightness.last_light_level = "normal" # static variable
```
Maybe you can give me a hint, or maybe I overcomplicate matters, as the raspicam V3 would regulate the brightness on its own?

davidplowman, 2025-08-06:
The "lux" metadata value is supposed to be (as far as possible) an exposure-independent measure of the absolute scene brightness. So for example, if the scene is fixed, you can set a high or low exposure time, but the lux value should basically stay the same. So it should generally be a good number for judging whether you are dealing with a dark or bright scene. It is derived from the exposure time, gain, average pixel value and compared to reference values obtained during initial camera calibration.

Normally, one would just leave the AGC/AEC algorithms running (i.e. don't set fixed exposure/gain values), and the camera system will regulate the exposure time and gain to give you a reasonable image. This in-built algorithm is both very flexible and highly customisable, and can probably be made to do almost anything you want. You could change anything from the "exposure value" (EV) using a runtime control, to the detailed under-the-hood parameters in the camera tuning file. There is documentation on this algorithm in the [Camera Tuning Guide](https://datasheets.raspberrypi.com/camera/raspberry-pi-camera-guide.pdf) - section 5,9 - and I don't mind helping if a fresh issue is opened here.

The first thing would be to determine in what way the current AGC/AEC metering doesn't meet your requirements - is that something you could explain?

herbbetz, 2025-08-06:
Many thanks for your hints, I also did not know about 'https://datasheets.raspberrypi.com/'. Concerning, why the current AGC/AEC metering would not meet my requirements: It all started with yellowish camera recordings under an indoor light source and chatGPT proposing the following code for automatic white balance freezing at the program's start:
```
        picam.set_controls({"AwbEnable": True})
        time.sleep(1.5)
        gains = tuple(round(g, 2) for g in picam.capture_metadata()["ColourGains"])
        camsetting = {
            "AeEnable": False,
            "AnalogueGain": 1.5,
            "ExposureTime": 15000,
            "AwbEnable": False,
            "ColourGains": gains
        }
        picam.set_controls(camsetting)

```
AeEnable is by the way not supported with the cheap Raspicam v3 anyway (RPi 4B, bookworm). 

Now I realized, I could also manipulate exposure and gain, and I hoped, I would benefit from this in more extreme outdoor light settings (sunset, morning sun backlight). From your response I now gather, that I just need to rely on the algorithm built into picamera2. I will monitor this for some time to get more experience. I changed my above function accordingly:
```
def get_brightness(picam, camsetting, now):
    metadata = picam.capture_metadata()
    # luxdata = metadata.copy()
    metalux = round(metadata.get("Lux")) # or metadata["Lux"], metadata.get("Lux", None)
    ...
```
One final question, to small for an own thread: when exiting a picamera2 script, the Rpi cam is normally released, so there is no code necessary like `if picam.running: picam.stop()` or `picam.stop_recording`? Is there an attribute like `picam.running`?

herbbetz, 2025-08-07:
Sorry to bother again. Just as I started watching the built-in AGC/AEC metering (and no longer manipulating it by my own `picam.set_controls(camsetting)`), something strange happened this morning, which has already been described [here](https://github.com/raspberrypi/picamera2/issues/1060). Yesterday (08-06) the raspicam V3 (RPi4B, bookworm) ran until darkness, when I have shutdown the system. It was rebooted this morning (08-07) at 6:00 by a smartplug and still kept the exposure/gain of the evening before, which resulted in very overexposed videos. A reboot at 7:45 resetted it to normal. Watch the following graphs: 

[image]
[image]
[image]