raspberrypi/picamera2 #1312: [HOW-TO] remove the first frame count in a rolling shutter camea

almowery, 2025-08-24:
I have been trying to use indi_pylibcamera which is built on picamera2

Using rpicam-still I have been able to achieve getting one frame image taking.  Is there a mode or settings to do this in Picamera2?

rpicam-still -o test.raw --timeout 1 --tuning-file imx283.json --analoggain 1 --awbgains 1.0,3.5 --saturation 1
--contrast 1 --sharpness 1 --brightness 0 --nopreview --autofocus-mode manual --immediate --gain 1 --shutter 10000000

When using the rolling shutter camera as an astro camera, taking several frames repeatedly, there is no need for an initial frame to dump the sensor image.
this script runs the rolling shutter camera without taking extra frames as outlined  
I originally got rpicam to work with --immediate because this was fixed and outlined here.
https://github.com/raspberrypi/rpicam-apps/issues/524

Looking at the picamera2 code I don't see the immediate option listed as an available option is that true?  Do you know if it can be added using the cam-helper file?
Or anyway for me to make my own special build?

Any advice on this would be very helpful.  Thank You.

davidplowman, 2025-08-24:
Picamera2 generally doesn't have all the options you see in (for example) rpicam-still, simply because it doesn't need them. You can write a script that gives you the exact behaviour you want - it's not like a non-interactive C++ executable where these things have to be specifically implemented.

In the case for your example command, you want something like
```
from picamera2 import Picamera2

tuning = Picamera2.load_tuning_file("imx283.json")
picam2 = Picamera2(tuning=tuning)
controls = {'AnalogueGain': 1, 'ColourGains': (1.0, 3.5), 'ExposureTime': 10000000}
config = picam2.create_still_configuration(controls=controls)
picam2.configure(config)
picam2.start()
picam2.capture_file("test.jpg")
```
This should capture the very first camera frame, like the `--immediate` option (but untested - don't currently have a Pi in front of me).

A few points:
* I didn't list all your control values - most of them are the defaults.
* The `-o` option in rpicam-still will save a JPEG, not a raw or DNG file.
* Can't help you otherwise the indi-pylibcamera, I have never heard of it.

davidplowman, 2025-08-29:
I think this question has now cropped up [here](https://github.com/raspberrypi/rpicam-apps/issues/856).

I'll close this one so as to keep the discussion in one place.