raspberrypi/picamera2 #1322: [HOW-TO] Auto-Exposure at night

mmoollllee, 2025-09-22:
I'm trying to set up capturing images 24/7 with proper exposure at night. I'm using a RPi HQ Camera with an RPi 3A+ and most recent PiCamera2,...

With the following code, the longest exposure time I get is '66_657', while I expect it to be '2_000_000'. It works if I set this manually. Am I doing something wrong? I've been trying to disable custom metering and constraint, but it didn't help.

```
def capture(self):
   timeofday = get_timeofday() # "day", "dusk", "night", "dawn", "morning", "evening"

   tuning = Picamera2.load_tuning_file("imx477.json")
   algo = Picamera2.find_tuning_algo(tuning, "rpi.agc")

   transform = Transform(hflip=True, vflip=True) if self.rotate == 180 else Transform()

   # custom metering, to define where the camera should measure exposure
   metering = [5, 5, 5, 10, 0, 2, 2, 1, 2, 5, 0, 5, 5, 0, 0]
   # custom constraint, because the original "shadow" constraint is too aggressive
   custom_constraint = [{
         "bound": "LOWER",
         "q_lo": 0.20, "q_hi": 0.50,
         "y_target": [
            0,    0.20,
            1000, 0.15
         ]
      }]
   # custom long exposure settings (will be used if timeofday != "day").
   long_exposure = {
      "shutter": [500_000, 1_000_000, 2_000_000, 4_000_000, 8_000_000, 10_000_000],  # 0.5–10 s
      "gain": [ 1.0, 1.5, 2.0, 4.0, 6.0, 12.0 ]
   }

   # apply custom settings to all channels
   for channel in algo["channels"]:
      if "metering_modes" not in channel or not isinstance(channel["metering_modes"], dict):
         channel["metering_modes"] = {}
      channel["metering_modes"]["custom"] = {"weights": metering}

      if "constraint_modes" not in channel or not isinstance(channel["constraint_modes"], dict):
         channel["constraint_modes"] = {}
      channel["constraint_modes"]["custom"] = custom_constraint

      channel["exposure_modes"]["long"] = long_exposure

   self.picam2 = Picamera2(tuning=tuning)

   self.picam2.start(show_preview=False)

   # default controls
   camera_ctrls = {
      "AeConstraintMode": controls.AeConstraintModeEnum.Custom,
      "AeMeteringMode": controls.AeMeteringModeEnum.Custom,
      "AeExposureMode": controls.AeExposureModeEnum.Short,
      "FrameDurationLimits": (100, 1000000),
   }

   # adjust controls for night/dusk/dawn/...
   if timeofday != "day":
      camera_ctrls.update({
         "AeExposureMode": controls.AeExposureModeEnum.Long,
         "FrameDurationLimits": (500_000, 15_000_000),
         "AeFlickerMode": controls.AeFlickerModeEnum.Off, # make sure this is not interfering
         # "ExposureTime": 2_000_000, # what I'd expect
      })

   self.picam2.set_controls(camera_ctrls)

   capture_config = self.picam2.create_still_configuration(controls=camera_ctrls, transform=transform)
   self.picam2.switch_mode(capture_config)

   # some camera cycles to settle exposure
   meta = None
   for _ in range(3):
      meta = self.picam2.capture_metadata()
      time.sleep(0.25)

   self.picam2.capture_file(self.filepath)
   with open(self.filepath.replace(".jpg", ".json"), "w") as f:
         json.dump(meta, f, indent=2)

```

mmoollllee, 2025-09-22:
Awesome! Thanks for the fast reply!

This explains why I remembered it was working before (but never shipped to any production device). So now I might want to update this for older libcamera versions (where this would actually work without the workaround), and to current versions (where the workaround is needed).
Maybe you can tell me if this will this work for both (current and older)? I'm afraid of updating anything, because I have the feeling, that I'm running into memory issues more often on a RPi 3A+ (with 512 MB RAM) with the current libcamera, compared to older versions...

davidplowman, 2025-09-22:
I think the workaround version should work for older non-broken versions of libcamera as well.

Some forensic archaeology suggests it got broken around January 2025, and the code before that ignored the enabled/disabled status entirely (as does the fix I have posted upstream to libcamera), so my expectation is that it would be OK.

mmoollllee, 2025-09-22:
Works on both versions! 👌

```
# new
rpicam-hello --version
rpicam-apps build: v1.9.0 a819af48889f-dirty 12-09-2025 (07:56:45)
rpicam-apps capabilites: egl:1 qt:1 drm:1 libav:1
libcamera build: v0.5.2+125-69c6546d-dirty (2025-09-12T07:42:08BST)

# old
libcamera-hello --version
rpicam-apps build: v1.6.0 9825703b669a-dirty 05-02-2025 (06:33:05)
rpicam-apps capabilites: egl:1 qt:1 drm:1 libav:1
libcamera build: v0.4.0+78-818d45eb-dirty (2025-02-14T02:13:36GMT)
```

davidplowman, 2025-11-11:
I'm afraid it probably does. I've looked again at the code to ensure our "next" branch is fixed, so the problem should go away after we issue our next release.

davidplowman, 2025-11-28:
No, because it's a libcamera bug, not a Picamera2 bug. We've haven't formally released 0.3.32 into apt yet, but when we do, which is probably just a day or two away, we'll ship an updated libcamera as well which _should_ resolve it. So just hang tight for a few more days!

HyperLethalVector, 2025-12-17:
Ding!

Sorry to necro, but I figured my issue sits closer to this issue.

I have an arducam hawkeye 64mp camera, 

```
import time
from picamera2 import Picamera2
import libcamera

tuning = Picamera2.load_tuning_file("arducam_64mp.json")
algo = Picamera2.find_tuning_algo(tuning, "rpi.agc")
long_exposure = {"shutter": [500_000, 1_000_000, 2_000_000, 4_000_000, 8_000_000, 10_000_000],
                 "gain": [ 1.0, 1.0, 1.0, 1.0, 1.0, 1.0 ]}
channel = algo["channels"][0]
channel["exposure_modes"]["long"] = long_exposure
channel["exposure_modes"]["short"] = long_exposure
channel["exposure_modes"]["normal"] = long_exposure
picam2 = Picamera2(tuning=tuning)
camera_ctrls = {"FrameDurationLimits": (1_000, 15_000_000)}
config = picam2.create_still_configuration(controls=camera_ctrls)
picam2.configure(config)
picam2.start()
time.sleep(5)

picam2.set_controls({"AeEnable": False, "AeExposureMode": libcamera.controls.AeExposureModeEnum.Long}) 
for _ in range(2):
    metadata = picam2.capture_metadata()
    exposure_time = metadata.get("ExposureTime", 0)
    analogue_gain = metadata.get("AnalogueGain", 0)
    print(f"ExposureTime: {exposure_time / 1_000_000:.3f} s, AnalogueGain: {analogue_gain}")
    time.sleep(1)
picam2.set_controls({"AeEnable": True})
time.sleep(5)
for _ in range(2):
    metadata = picam2.capture_metadata()
    exposure_time = metadata.get("ExposureTime", 0)
    analogue_gain = metadata.get("AnalogueGain", 0)
    print(f"ExposureTime: {exposure_time / 1_000_000:.3f} s, AnalogueGain: {analogue_gain}")
    time.sleep(1)

time.sleep(20)
```
Doing this prints the outputs: 
```
[2:47:26.327332992] [3167]  INFO Camera camera_manager.cpp:340 libcamera v0.6.0+rpt20251202+26-177de571-dirty (2025-12-03T09:12:19GMT)
[2:47:26.335953621] [3170]  INFO RPI pisp.cpp:720 libpisp version 1.3.0
[2:47:26.390646269] [3170]  INFO IPAProxy ipa_proxy.cpp:180 Using tuning file /tmp/tmp8bqx4ods
[2:47:26.395888621] [3170]  INFO Camera camera_manager.cpp:223 Adding camera '/base/axi/pcie@1000120000/rp1/i2c@88000/arducam_64mp@1a' for pipeline handler rpi/pisp
[2:47:26.395921066] [3170]  INFO RPI pisp.cpp:1182 Registered camera /base/axi/pcie@1000120000/rp1/i2c@88000/arducam_64mp@1a to CFE device /dev/media1 and ISP device /dev/media0 using PiSP variant BCM2712_D0
[2:47:26.399151343] [3167]  INFO Camera camera.cpp:1215 configuring streams: (0) 9152x6944-BGR888/sRGB (1) 9152x6944-RGGB_PISP_COMP1/RAW
[2:47:26.399248862] [3170]  INFO RPI pisp.cpp:1486 Sensor: /base/axi/pcie@1000120000/rp1/i2c@88000/arducam_64mp@1a - Selected sensor format: 9152x6944-SRGGB10_1X10/RAW - Selected CFE format: 9152x6944-PC1R/RAW
ExposureTime: 2.500 s, AnalogueGain: 1.0
ExposureTime: 2.500 s, AnalogueGain: 1.0
ExposureTime: 2.500 s, AnalogueGain: 1.0
ExposureTime: 2.500 s, AnalogueGain: 1.0
```
Which if I've put the lens in an almost pitch-dark room, I expect it to hit the upper limit?

Prior versions of libcamera I was able to reach that upper 15 second limit under the same conditions
(I'll have to dig up what version that was)

Any ideas what I'm doing wrong? 

attached the tuning file

[arducam_64mp.json](https://github.com/user-attachments/files/24202611/arducam_64mp.json)
(Sorry for the edits, I suck at code commenting)

davidplowman, 2025-12-17:
@HyperLethalVector Hi, I wasn't 100% clear exactly what your script was doing, but I wonder if the main problem is that you're not waiting for the controls to take effect? The following simple script seems to work fine for me, and I see very long exposures when I cover the camera. 
```python
from picamera2 import Picamera2
import libcamera

tuning = Picamera2.load_tuning_file("imx500.json")  # don't have any Arducam cameras
algo = Picamera2.find_tuning_algo(tuning, "rpi.agc")
long_exposure = {"shutter": [5_000, 10_000_000], "gain": [1.0, 1.0]}
channel = algo["channels"][0]
channel["exposure_modes"]["long"] = long_exposure

picam2 = Picamera2(tuning=tuning)

camera_ctrls = {"FrameDurationLimits": (1_000, 15_000_000)}
                "AeExposureMode": libcamera.controls.AeExposureModeEnum.Long}
config = picam2.create_still_configuration(controls=camera_ctrls, buffer_count=2)
picam2.configure(config)

picam2.start()

while True:
    metadata = picam2.capture_metadata()
    print(metadata['ExposureTime'])
```
Please make sure you have the latest camera software installed from the Raspberry Pi repositories. If you're using customised software from the camera module vendor, and find different behaviour from the standard Raspberry Pi distribution, then you may need to go back to the vendor for support.